sadr0b0t/arduino-timer-api #9: Сбивается счетчик количества импульсов (записанные значения в EEPROM)

YarD17, 2018-03-14:
Если отправить "1" в SerialPort то выдаст значения счетчика,которые сбрасываются при пере подключении к порту, хотя должны накапливаться.Плата Arduino UNO R3. 
Пример сбрасывания EEPROM:
3 3 2 45 163 146 
6 6 5 0 163 146 
9 9 8 0 163 146 
11 11 10 0 0 146 
13 13 12 1 0 146 
15 15 14 1 0 0 
17 17 16 1 0 0 
0 21 20 2 0 0

sadr0b0t, 2018-03-14:
У меня сейчас под рукой нет классической ардуины Uno R3, но я проверил такой простой скетч на платах Arduino Leonardo, Arduino Due (SAM), ChipKIT Uno32 (клон ардуино с чипом PIC32)

~~~cpp
void setup() {
  Serial.begin(9600);
  pinMode(13, OUTPUT);
}

int counter = 0;

void loop() {
  counter++;
  Serial.println(counter);
  delay(1000);
}
~~~

Шьем плату, открываем Searial monitor, ждем, пока счетчик немного посчитает, закрываем Serial monitor, опять открываем Serial monitor, смотрим, откуда продолжается счет.

У меня:
Arduino Leonardo: счет продолжается (плата НЕ перезагружается при закрытии/открытии окна Serial monitor)
Arduino Due, ChipKIT Uno32: счет начитается заново (плата перезагружается при закрытии/открытии окна Serial monitor)

Это точно не ваша проблема?

YarD17, 2018-03-14:
counter++; - запись в оперативную память работает, но сбрасывается при закрытии/открытии окна Serial monitor
А запись в EEPROM.write(1,counter) - Периодически сбрасывается во время работы открытого окна Serial monitor и при закрытии/открытии окна.
Возможно, у меня, как в случае с  "Uno32: счет начитается заново (плата перезагружается при закрытии/открытии окна Serial monitor)"

sadr0b0t, 2018-03-14:
посмотрите ссылки
https://arduino.stackexchange.com/questions/439/why-does-starting-the-serial-monitor-restart-the-sketch
https://playground.arduino.cc/Main/DisablingAutoResetOnSerialConnection